stakwork/sphinx-key #61: Dockerize broker and test with CLN

Evanfeenstra, 2022-09-08:
lets figure out how to run "sphinx CLN" (with broker instead of hsmd) inside docker. 

Here is a CLN setup that Tobi helped us with https://github.com/stakwork/sphinx-stack/blob/master/cln/cln.yml

Evanfeenstra, 2022-09-14:
broker is building in Docker now (`docker build -f broker/Dockerfile -t sphinx-key-broker .`)

@omoniyi24 do you think you can run CLN in docker-compose with the broker as the signer daemon?

There is a `subdaemon` command to switch out default daemons for another one. like: `subdaemon=hsmd:PATH_TO_BROKER_EXECUTABLE`. But i am not sure how to share executables between docker containers?

https://lightning.readthedocs.io/lightningd-config.5.html

Evanfeenstra, 2022-09-17:
Here's a CLN with broker... https://github.com/stakwork/lightning/tree/sphinx-cln

check out the `cln-grpc` stuff here https://github.com/stakwork/lightning/blob/sphinx-cln/plugins/Makefile, probably we need to add broker there as well?

omoniyi24, 2022-09-20:
@Evanfeenstra I couldn't build the broker image with `docker build -f broker/Dockerfile -t sphinx-key-broker .` It throws an error `#10 0.323 error: toolchain 'nightly-x86_64-unknown-linux-gnu' is not installed`. Do I need to install any dependency before building.

Evanfeenstra, 2022-09-20:
trying a new approach of just injecting the broker built on the same image as CLN (debian:bullseye-slim). the executable can be extracted like this:

```
cid=$(docker create sphinx-key-broker)
docker cp $cid:/usr/src/sphinx-key-broker - > local-key-broker
docker rm -v $cid
```

closed 2022-10-06 by Evanfeenstra